rust-embedded-community/usb-device #1: Issues with Cell/RefCell using app!() macro

jamwaffles, 2018-09-01:
I have a project using `usb-device` (and [`stm32f103xx-usb`](https://github.com/mvirkkunen/stm32f103xx-usb)) which worked great until I added the `app!()` macro from RTFM. Full example can be found [here](https://github.com/jamwaffles/blue-pill-logic-analyser/blob/master/blue-pill-logic-analyser-firmware/src/main.rs). Try cloning the repo, run `cargo build -p blue-pill-logic-analyser-firmware` to see the following spew of errors:

```
error[E0277]: `core::cell::RefCell<stm32f103xx_usb::regs::PacketMemory>` cannot be shared between threads safely
  --> blue-pill-logic-analyser-firmware/src/main.rs:28:1
   |
28 | / app! {
29 | |     device: hal::stm32f103xx,
30 | |
31 | |     resources: {
...  |
50 | |     },
51 | | }
   | |_^ `core::cell::RefCell<stm32f103xx_usb::regs::PacketMemory>` cannot be shared between threads safely
   |
   = help: within `stm32f103xx_usb::UsbBus`, the trait `core::marker::Sync` is not implemented for `core::cell::RefCell<stm32f103xx_usb::regs::PacketMemory>`
   = note: required because it appears within the type `stm32f103xx_usb::UsbBus`
   = note: required because of the requirements on the impl of `core::marker::Send` for `&'static stm32f103xx_usb::UsbBus`
   = note: required because it appears within the type `usb_device::endpoint::Endpoint<'static, stm32f103xx_usb::UsbBus, usb_device::endpoint::In>`
   = note: required because it appears within the type `cdc::SerialPort<'static, stm32f103xx_usb::UsbBus>`

error[E0277]: `core::cell::Cell<core::option::Option<usize>>` cannot be shared between threads safely
  --> blue-pill-logic-analyser-firmware/src/main.rs:28:1
   |
28 | / app! {
29 | |     device: hal::stm32f103xx,
30 | |
31 | |     resources: {
...  |
50 | |     },
51 | | }
   | |_^ `core::cell::Cell<core::option::Option<usize>>` cannot be shared between threads safely
   |
   = help: within `stm32f103xx_usb::UsbBus`, the trait `core::marker::Sync` is not implemented for `core::cell::Cell<core::option::Option<usize>>`
   = note: required because it appears within the type `stm32f103xx_usb::UsbBus`
   = note: required because of the requirements on the impl of `core::marker::Send` for `&'static stm32f103xx_usb::UsbBus`
   = note: required because it appears within the type `usb_device::endpoint::Endpoint<'static, stm32f103xx_usb::UsbBus, usb_device::endpoint::In>`
   = note: required because it appears within the type `cdc::SerialPort<'static, stm32f103xx_usb::UsbBus>`

error[E0277]: `core::cell::RefCell<[stm32f103xx_usb::bus::EndpointRecord; 8]>` cannot be shared between threads safely
  --> blue-pill-logic-analyser-firmware/src/main.rs:28:1
   |
28 | / app! {
29 | |     device: hal::stm32f103xx,
30 | |
31 | |     resources: {
...  |
50 | |     },
51 | | }
   | |_^ `core::cell::RefCell<[stm32f103xx_usb::bus::EndpointRecord; 8]>` cannot be shared between threads safely
   |
   = help: within `stm32f103xx_usb::UsbBus`, the trait `core::marker::Sync` is not implemented for `core::cell::RefCell<[stm32f103xx_usb::bus::EndpointRecord; 8]>`
   = note: required because it appears within the type `stm32f103xx_usb::UsbBus`
   = note: required because of the requirements on the impl of `core::marker::Send` for `&'static stm32f103xx_usb::UsbBus`
   = note: required because it appears within the type `usb_device::endpoint::Endpoint<'static, stm32f103xx_usb::UsbBus, usb_device::endpoint::In>`
   = note: required because it appears within the type `cdc::SerialPort<'static, stm32f103xx_usb::UsbBus>`

error[E0277]: `core::cell::Cell<u8>` cannot be shared between threads safely
  --> blue-pill-logic-analyser-firmware/src/main.rs:28:1
   |
28 | / app! {
29 | |     device: hal::stm32f103xx,
30 | |
31 | |     resources: {
...  |
50 | |     },
51 | | }
   | |_^ `core::cell::Cell<u8>` cannot be shared between threads safely
   |
   = help: within `stm32f103xx_usb::UsbBus`, the trait `core::marker::Sync` is not implemented for `core::cell::Cell<u8>`
   = note: required because it appears within the type `usb_device::bus::UsbAllocatorState`
   = note: required because it appears within the type `stm32f103xx_usb::UsbBus`
   = note: required because of the requirements on the impl of `core::marker::Send` for `&'static stm32f103xx_usb::UsbBus`
   = note: required because it appears within the type `usb_device::endpoint::Endpoint<'static, stm32f103xx_usb::UsbBus, usb_device::endpoint::In>`
   = note: required because it appears within the type `cdc::SerialPort<'static, stm32f103xx_usb::UsbBus>`

error[E0277]: `*const ()` cannot be shared between threads safely
  --> blue-pill-logic-analyser-firmware/src/main.rs:28:1
   |
28 | / app! {
29 | |     device: hal::stm32f103xx,
30 | |
31 | |     resources: {
...  |
50 | |     },
51 | | }
   | |_^ `*const ()` cannot be shared between threads safely
   |
   = help: within `stm32f103xx_usb::UsbBus`, the trait `core::marker::Sync` is not implemented for `*const ()`
   = note: required because it appears within the type `core::marker::PhantomData<*const ()>`
   = note: required because it appears within the type `hal::stm32f103xx::USB`
   = note: required because it appears within the type `stm32f103xx_usb::UsbBus`
   = note: required because of the requirements on the impl of `core::marker::Send` for `&'static stm32f103xx_usb::UsbBus`
   = note: required because it appears within the type `usb_device::endpoint::Endpoint<'static, stm32f103xx_usb::UsbBus, usb_device::endpoint::In>`
   = note: required because it appears within the type `cdc::SerialPort<'static, stm32f103xx_usb::UsbBus>`

error: aborting due to 5 previous errors
```

What could the types be changed to so that `app!()` can be used?

Thanks for all the hard work on making USB an easy possibility on the STM32F103 chip!

mvirkkunen, 2018-09-02:
These crates aren't quite ready to be used with interrupts without completely locking everything yet, because there are a couple of racy operations that I haven't been able to remove yet. It should be possible though, and definitely a design-goal.

I'm not sure if it's possible to use these at all with RTFM yet, because it seems to require type-level guarantees of correct concurrent behavior (which is a good thing).

The current task I'm working on is restructuring things so that they can be made interrupt-safe with minimal locking (some locking will probably always be involved), but that is going to take some time and possibly some slight API restructuring to be able to put in those type level guarantees.

Thanks for your interest, I'll leave this issue open because it's a good reminder, and I'll keep you posted!

mvirkkunen, 2018-11-30:
I finally got around to making an initial interrupt-driven example. Tackling RTFM (the new version at least) is up next.

https://github.com/mvirkkunen/stm32f103xx-usb/blob/548729ae7dd6d09f21e57b36651601b80bc87bfb/examples/serial_interrupt.rs

mvirkkunen, 2019-05-12:
Everything seems to work with RTFM now.

https://github.com/mvirkkunen/stm32f103xx-usb/blob/master/examples/serial_rtfm.rs

closed 2019-05-12 by mvirkkunen