rust-embedded-community/usb-device #3: Consider adding an `EndpointAddress` helper type

wez, 2018-09-14:
In implementing USB for atsamd21 I found that it led to slightly clearer code and less human errors to wrap up the `u8` in a helper type:

https://github.com/wez/atsamd21-rs/blob/usb/hal/src/usb/bus.rs#L17-L65
(I've also included the code below in case I force push that branch later)

This made is easier to ensure that I had the right values and were considering the cases correctly, vs. just using bitmasks and hoping that I got the sense of the directions correct.

```rust
#[derive(Debug, Clone, Copy)]
struct EndpointAddress(u8);

impl From<u8> for EndpointAddress {
    #[inline]
    fn from(addr: u8) -> EndpointAddress {
        EndpointAddress(addr)
    }
}

impl From<EndpointAddress> for u8 {
    #[inline]
    fn from(addr: EndpointAddress) -> u8 {
        addr.0
    }
}

impl EndpointAddress {
    const INBITS: u8 = EndpointDirection::In as u8;

    #[inline]
    fn from_parts(idx: usize, dir: EndpointDirection) -> Self {
        EndpointAddress(idx as u8 | dir as u8)
    }

    #[inline]
    fn direction(&self) -> EndpointDirection {
        if (self.0 & Self::INBITS) != 0 {
            EndpointDirection::In
        } else {
            EndpointDirection::Out
        }
    }

    #[inline]
    fn is_in(&self) -> bool {
        (self.0 & Self::INBITS) != 0
    }

    #[inline]
    fn is_out(&self) -> bool {
        (self.0 & Self::INBITS) == 0
    }

    #[inline]
    fn index(&self) -> usize {
        (self.0 & !Self::INBITS) as usize
    }
}
```

mvirkkunen, 2018-09-15:
I eventually ended up with something similar but not quite as type safe here:

https://github.com/mvirkkunen/stm32f103xx-usb/blob/master/src/bus.rs#L30

Another idea would be to pass around references to the entire Endpoint struct. Then you'd have to have separate methods for handling in/out endpoints - but at least in the STM32F103 implementation every method that takes an endpoint address starts by either validating the direction (which wouldn't be necessary with separate types) or branching based on it (which would become two different methods). Hmm. I wonder if it's the same on other platforms as well?

mvirkkunen, 2018-09-17:
Added this pretty much as-is in 60db7e175e6dcea0b9c0f6b56564d1253bf0fd90. It made my implementation simpler too, and it should help with other platforms as well. Thanks.

closed 2018-09-17 by mvirkkunen