rust-embedded-community/usb-device #5: Consider an alloc_ep API change

wez, 2018-09-16:
This function serves multiple purposes today:

1. To find an available endpoint (when `addr` is specified as `None`)
2. To allocate buffer space for the endpoint
3. To record configuration that is subsequently applied during `reset`

Looking at implementing this for a second device, I'm looking at duplicating code for 1) and 2) that are not really implementation specific.

I'd like to propose that the `addr` argument be changed to no longer be optional; in other words that the usb device keep track of which endpoints it has configured so that the bus can be simplified to validating whether the requested arguments are valid.

Separately, I'd also like to hear your thoughts on buffer allocation.  Both the ATSAM and NRF USB implementations allow 64 byte buffers per endpoint (with more for isochronous endpoints, but those seem better suited for DMA usage).  Both of these could use a simple bump allocator with some static storage.  I think this might also be true for your STM implementation but I'm not super sure.  Does it make sense to include a struct in the usb_device crate that does something like this to save on duplicated implementation efforts?
https://github.com/wez/atsamd21-rs/blob/master/hal/src/usb/bus.rs#L189-L232
Devices that work differently wouldn't be required to use it.

For point 3) it would certainly be convenient for me if usb device kept track of the allocated endpoints and passed that set into `reset`.  That would allow cutting out some additional state tracking from my bus implementations.  I do recognize that I've only really looked at two different implementations and may be overly narrowly focused.

Overall, what I'm proposing is that:

* the parameters for `alloc_ep` be turned into a struct (name, maybe `EndpointConfig`?)
* have the `alloc` call manage the `Option<u8>` resolution to find an available slot
* pass the parameters into `alloc_ep` and have the primary purpose of `alloc_ep` be to validate that it will work with the hardware and allocate the buffer, returning a filled in `EndpointConfig` that also holds the allocated buffer address on success, or an error on failure
* the `reset` call receives a vec/slice or `EndpointConfig` instances previously allocated so that the bus can apply that configuration.

mvirkkunen, 2018-09-16:
The reason endpoint allocation and tracking is completely delegated to UsbBus is that it seems vary a lot per platform. For example on the STM32F103 there's the limitation that endpoints always come in pairs or IN/OUT endpoints which must have the same type (Interrupt, Bulk, etc) - even if the other one goes unused. Additionally bulk and isochronous endpoints allow for double-buffering which uses up an entire endpoint pair per direction, and needs two buffer addresses (there's no API for requesting this behavior yet, but it'd be nice to be able to support that). I've seen other restrictions on other platforms.

The memory allocation is also somewhat bizarre due to how the packet buffer RAM is implemented (it's a dual-port RAM for 8-bit data, arranged in 16-bit word, where each word is accessed via 32bit accesses). There are also devices that don't even let you write to packet memory directly via memory access, but you must go through a fixed FIFO register, so you don't even really get a memory address for the buffer.

The `Option<u8>` for the endpoint address does feel like a wart - it's really only used for allocating endpoint 0 currently. But it could also be used by a `UsbClass` to allocate an endpoint with a fixed, known address, in order to support an existing host driver that for some reason expects an endpoint with a fixed address. Changing the arguments into a struct could make sense because the number of arguments is getting a bit high.

Due to how the rules for endpoint allocation seem to be pretty different across devices, I'm hesitant to add too much logic for it in the device-agnostic crate.

wez, 2018-09-18:
Ok, that all makes sense.  Thanks for sharing the context!  In light of all of that, I withdraw this issue.  I think it may still be useful to consolidate/centralize basic buffer allocation, but that's not the biggest thing here.

closed 2018-09-18 by wez

mvirkkunen, 2018-09-18:
Optional reusable helper types to make implementing UsbBus easier could still be included. However it might be hard to judge what actually is reusable before there are a few more full implementations. If many implementations end up re-implementing the exact same logic, it could be extracted into this crate.