rust-embedded-community/usb-device #6: Write a test suite for real hardware

mvirkkunen, 2018-09-19:
While `UsbDevice` itself can be tested with standard unit tests by using a mock `UsbBus`, there also needs to be a hardware test suite for testing platform-specific `UsbBus` implementations. The test suite could consist of:

* A test `UsbClass` which can exercise various transfer and endpoint types (put this in a crate)
* A skeleton program for initializing a `UsbDevice` for testing, that can be expanded into a full test program that can be flashed to a target by adding the platform-specific parts
* A host side program written using `libusb` that talks to the device and tests everything

It's not quite as nice to run as unit tests, as one needs the actual hardware to run on, but I don't think it's feasible to try to write pure unit tests hardware register manipulation.

This should probably be implemented after the API is more stable, but I'm opening this issue so I won't forget.

nickray, 2018-10-19:
I would like to use this crate for my SoloKeys project (open source FIDO2 plus potentially more, like ssh, gpg, wireguard), so once my Rust is up to speed, would be interested in helping with this for the STM32L4x2 chips.

mvirkkunen, 2018-12-19:
TestClass and test_class_host now have a basic test suite. Could be expanded later but it already exercises a bunch of edge cases.

closed 2018-12-19 by mvirkkunen