rust-embedded-community/usb-device #9: Design: Figure out a better way to keep track of the list of classes

mvirkkunen, 2019-05-12:
`UsbDevice::poll` currently has a nasty warning:

```
/// Note: The list of classes passed in must be the same for every call while the device is
/// configured, or the device may enumerate incorrectly or otherwise misbehave. The easiest way
/// to do this is to call the `poll` method in only one place in your code, as follows:
```

Things will break in weird and wonderful ways if the user doesn't always pass in the same slice of classes in the same order.

What I would really want is for UsbDevice to hold references to the list of classes. However this is a problem because both UsbDevice and the user would want to have a `&mut` to each class. Earlier versions used immutable references but this meant all classes had to be written using interior mutability which was nasty.

Possible solutions:

1. Use RefCells. This means the user has to borrow_mut every time they want to use their serial port or whatever though. Not ideal.
2. Use something similar to smoltcp where they have a SocketSet type you pass to poll, and instead of the actual sockets you get socket handles you can then request access to via the SocketSet. This is probably even more convoluted to use  the RefCells though. ([Example](https://github.com/m-labs/smoltcp/blob/185a0ca7811cb7385cf26a7627a7e0d8c357d147/examples/ping.rs#L113))
3. Live with it

ianrrees, 2021-03-11:
I like solution 2, because it helps fix a problem in the no-scheduler case:

Currently, the classes reference back to the UsbBus via endpoints, which at least in some cases means the UsbBus implementation needs something like a mutex ([ATSAMD example](https://github.com/atsamd-rs/atsamd/blob/a78ba161be13d8cb5aa40f177e3e8d31e6b8d4d5/hal/src/samd21/usb/bus.rs#L207-L209)) to prevent multiple "concurrent" accesses to the bus.  The mutex is problematic when there's no scheduler, because locking it disables all interrupts for the duration of the lock.  A mutex with a `try_lock()` could be implemented which doesn't block if the mutex is held already, returning an error instead, and which only disables all interrupts for a couple instructions while locking/unlocking.  But, if the USB classes and bus were all owned by one struct, then I think we could get rid of the mutex requirement since Rust would ensure there's only one execution context that can access an instance of that struct.

FeldrinH, 2025-07-12:
It seems to me that solution 2 has another benefit for non-Sync UsbBus implementations: if UsbDevice owns all the classes then it could take ownership of UsbBus on creation, which would allow UsbDevice to be Send even if UsbBus isn't Sync.