rust-embedded-community/usb-device #14: SetConfiguration cannot deconfigured the device

lichtfeind, 2019-05-23:
I am porting the USB device driver for the STM32F042 and i am testing it with the [USBCV20] tool.

While doing so i learned that SetConfiguration() with zero as value should set the device back into Addressed state(i am guessing that from the tools output).
I did not find a clear description of this in the USB Specification(Revision 2.0) except in Figure 9-1(Device State Diagram) on Page 240 where it is indicated that is possible to "Deconfigured" a device.

I made a small patch that turns some more tests green for me.
[https://github.com/lichtfeind/usb-device/commit/4a5076ac28f9b23bf84d084e0c397726e4acce4e](https://github.com/lichtfeind/usb-device/commit/4a5076ac28f9b23bf84d084e0c397726e4acce4e)

I am unsure how to proceed with this as it might just get fixed when multiple configurations get supported.
So i just wanted to put it here in case someone knows more about it and can tell me were its specified.

[USBCV20]: https://www.usb.org/usb2tools

mvirkkunen, 2019-05-23:
In addition to that diagram the SetConfiguration request in the spec also states:

> **Configured state:** If the specified configuration value is zero, then the device enters the Address state.
> If the specified configuration value matches the configuration value from a
> configuration descriptor, then that configuration is selected and the device remains in
> the Configured state. Otherwise, the device responds with a Request Error.

I guess supporting that bit doesn't hurt, although there really isn't any way to do multiple configurations currently. AFAIK very few devices ever use this. Multiple interface alternate settings would be useful though, but I'm not entirely sure how to do those either.

Looking at that state diagram it seems I've also implemented the Suspended state incorrectly, as it seems the device should be able to return to other states besides Default from it.

lichtfeind, 2019-05-23:
I am just trying to as complied as i can get. But it is not relay something that is imported to the function of the device as i have no clues as to what it is used for. Except in that protocol test to test if descriptors are accessible in configured and addressed state.

Suspend/Resume is also flagged red by that tool. But as of now that could be an error on my part as the device so fare does not like USB-Hubs. I suspect some signal issues, me doing something wrong or something else interfering. But i will add protocol to the list of possible causes and will look into it tomorrow.

lichtfeind, 2019-05-24:
The Suspend/Resume test passes with an STM32F103 BluePill so its a problem with my driver.

mvirkkunen, 2019-05-24:
I should probably try the official test tools out too!

lichtfeind, 2019-05-29:
I found the problem with Suspend/Resume.
https://github.com/lichtfeind/stm32f103xx-usb/commit/0437d580cbac86c503cc7a7276d6e0f1f4fc7b4b
The order in which this should happen is mention in the reference manual for the stm32f0x2 and stm32f103.
 - STM32F103: RM0091 Rev 9 Section: 30.5.5, Page: 874
 - STM32F0x2: RM0008 Rev 20 Section: 23.4.5, Page: 636

Both state to first clear FSUSP then read RXDP/RXDM.
Not doing so causes RXDP to appear high.

Now everything appears to be working.

It works for the stm32f103 without this patch so i am unsure if a pull request would be needed.

But i will late make a pull request for the SetConfiguration problem, so i can use upstream.

Disasm, 2019-06-02:
@lichtfeind I'm working on universal stm32 device-support library: https://github.com/Disasm/stm32-usbd
Seems like some of your changes would be quite useful for me. Do you mind contributing your changes there or can I just grub them and mention you somehow?

mvirkkunen, 2019-06-02:
I implemented the functionality for exiting the Configured state. There's still more work in this crate as well wrt correct state management and the Suspended state, but this is a good addition as it turns a lot of tests green in the official test tool.

Do you want to make a PR for stm32f103xx-usb with the suspend bit manipulation @lichtfeind ?

closed 2019-06-02 by mvirkkunen

lichtfeind, 2019-06-03:
Sure, but i am traveling until next Tuesday so i can't test it for the stm32f103 until then.

commit 9c3ffb96 (2019-06-02): Support SET_CONFIGURATION 0 to exit Configured state

Closes #14. Thanks to @lichtfeind for spotting this.