rust-embedded-community/usb-device #21: Design: New endpoint allocation/configuration API

mvirkkunen, 2019-07-29:
The current API with `UsbBusAllocator` is convoluted and does not support alternate settings. A new API is required for allocating endpoints as well as enabling them per interface alternate setting.

## Design goals

1. Easy to use.
2. Efficiently use the available endpoints even on devices that have special endpoint use restrictions.
3. Support interface alternate settings as specified in the USB specification.
4. Support allocating a fixed endpoint address for emulating devices with drivers that expect certain endpoint addresses.
5. Support some way to communicate enabling platform-specific features such as packet queue size (some peripherals support queueing multiple packets) or double-buffering

In addition to allocating endpoint addresses/registers/types, the allocator will allocate backing memory or FIFOs for endpoints on platforms that use them.

## Platform quirks

Some chips have restrictions on the usage of the endpoints they provide and the allocation API should try to use their resources as efficiently as possible.

### STM32 USB full speed (STM32F042, STM32F103, etc)

Endpoints come in pairs of IN and OUT that must have the same address index as well as the same transfer type. One register controls two endpoints. This means that if two endpoints in opposite directions are needed that are always the same transfer type, it's more efficient to allocate them as a pair.

### nRF52480

Endpoints have fixed addresses. Address indexes 1-7 can do bulk or interrupt transfers and index 8 can do isochronous transfers. Therefore the same endpoint address can never be e.g. isochronous in one alternate setting and bulk in another.

Only EP0 can do control transfers. Control transfers over other endpoints are probably within USB spec, but nobody does them in practice.

nickray, 2019-10-10:
Just for reference, the LPC55S69 has both a full speed and a high speed device peripheral. The former has 5 logical/10 physical, the latter (supposedly) has 6 logical/12 physical endpoints. Logical endpoint 0 can do only control, the higher endpoints can do interrupt/bulk/isochronous, optionally double-buffered. AFAIU the non-control endpoints can be individually allocated.

mvirkkunen, 2020-02-06:
Note: the "endpoint-trait" branch has some idea for a new allocator API

mvirkkunen, 2020-04-11:
And the branch now has an allocator system that should be able to work with all these quirks.

closed 2020-04-11 by mvirkkunen