rust-embedded-community/usb-device #26: Interface association

katyo, 2019-10-04:
I need join several classes in a single device, say two CDCACM and one HID.
To get it working in case of CDC I use USB interface association, which looks like:

```
Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       0
      iFunction               0
    Interface Descriptor: // interrupt
    Interface Descriptor: // bulk
```

But allocator doesn't provide a way to get the next allocated interface number nor the count of allocated interfaces. Also the allocator is not a trait now so it is not possible to implement proxy-allocator to create assotiation.

Because the interface association is a USB-wide spec (not CDC specific thing), I think this crate should care about support of this feature.
Of course, some host drivers (like linux) may work without interface association at all but in common case it is a wrong practice.

katyo, 2019-10-04:
I think turning allocator into trait may help to solve this issue, because it makes possible implement `UsbIfaceAssoc` like so:

```rust
pub struct UsbIfaceAssoc {
  first_interface: Cell<u8>,
  interface_count: Cell<u8>,
  // other fields
}

impl UsbClass for UsbIfaceAssoc {
  fn get_configuration_descriptors(&self, writer: &mut DescriptorWriter) -> Result<()> {
    // write association descriptor
  }
}

impl UsbBusAllocator for UsbIfaceAssoc {
  fn interface(&self) -> InterfaceNumber {
    // capture first interface number
    // and count allocated interfaces
  }
  // other required trait methods
}
```

mvirkkunen, 2019-10-04:
I've been thinking about this before and the idea I had was more like as follows:

1. When creating the `UsbDevice` the user specifies that they want a composite device, and that sets a bit in the device settings, as well as setting the class/subclass numbers to the "look at the IADs" values.
2. `DescriptorWriter` gains a new method called `interface_association_descriptor` that writes out the descriptor, taking as arguments the first interface number (which is inside an `InterfaceNumber` at that point), the number of interfaces and the class etc, and then writes out a descriptor. The class implementation calls this method in its own `get_configuration_descriptors` method. The method could be made inert (i.e. not do anything) if IADs are not requested, or there could be a method to check if they're needed.

How does that sound like to you?

This does allow classes to write out an invalid IAD, but then again the `DescriptorWriter` doesn't go out of its way to make sure descriptors are valid.

katyo, 2019-10-04:
Sounds pretty fine.
I think it completely solve this issue, but require little changes of `usbd-serial` crate too.

mvirkkunen, 2020-06-20:
An implementation for IADs exists in the endpoint-trait branch.

mvirkkunen, 2021-03-13:
Basic support was added in #42.

closed 2021-03-13 by mvirkkunen