rust-embedded-community/usb-device #35: USB 2.1 and BOS descriptors break things.

FrozenDroid, 2020-02-09:
Hi.

I saw that https://github.com/mvirkkunen/usb-device/pull/29 got merged.
It appears that this change is definitely breaking. My code wouldn't work, and once I changed the USB version to 2.0, so that BOS descriptors wouldn't get requested, it started working again. This happens on both macOS and Windows.
I think this has something to do with this line: https://github.com/mvirkkunen/usb-device/blob/master/src/descriptor.rs#L72

mvirkkunen, 2020-02-10:
Hi,

It seems there was a version dependent problem with that descriptor! The descriptor worked successfully with some versions of Windows for instance, but clearly not all of them. #34 adds a "dummy" descriptor which seems to make Windows and macOS happy. Could you verify that version 0.2.5 works for you before I close this? Thanks!

I've also yanked version 0.2.4 because it was broken for many users.

FrozenDroid, 2020-02-10:
Yep, that seems to have done it.  
I've only tested Windows right now, but I think it's save to assume it'll work in macOS too.  
Thanks!

closed 2020-02-10 by FrozenDroid