rust-embedded-community/usb-device #36: Rapid interface control_in fails with pipe errors / timeout

sameer, 2020-03-27:
Hello,

I'm implementing a [USB JTAG Blaster class](https://github.com/sameer/usbblaster-rs/blob/master/src/blaster/class.rs), and I've run into a problem with control_in.

There's an emulated ROM for [FT245](https://www.ftdichip.com/Support/Documents/DataSheets/ICs/DS_FT245R.pdf) read by a driver with vendor-type control_in requests. The index is used to specify the address in the ROM.

I have a test program to verify the emulated ROM contents match as expected:
```rust
use libusb::{Direction,RequestType,Recipient};
use std::time::Duration;

const ROM: [u8; 128] = [
    0x0, 0x0, 0xfb, 0x9, 0x1, 0x60, 0x0, 0x4, 0x80, 0xe1, 0x1c, 0x0, 0x0, 0x2, 0x94, 0xe, 0xa2,
    0x18, 0xba, 0x12, 0xe, 0x3, 0x41, 0x0, 0x6c, 0x0, 0x74, 0x0, 0x65, 0x0, 0x72, 0x0, 0x61, 0x0,
    0x18, 0x3, 0x55, 0x0, 0x53, 0x0, 0x42, 0x0, 0x2d, 0x0, 0x42, 0x0, 0x6c, 0x0, 0x61, 0x0, 0x73,
    0x0, 0x74, 0x0, 0x65, 0x0, 0x72, 0x0, 0x12, 0x3, 0x31, 0x0, 0x32, 0x0, 0x33, 0x0, 0x34, 0x0,
    0x35, 0x0, 0x36, 0x0, 0x37, 0x0, 0x38, 0x0, 0x2, 0x3, 0x1, 0x0, 0x52, 0x45, 0x56, 0x42, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0,
    0x0, 0x0, 0x0, 0x7a, 0x96,
];

fn main() {
    let mut ctx = libusb::Context::new().unwrap();
    let mut handle = ctx.open_device_with_vid_pid(0x09FB, 0x6001).unwrap();
    let mut buf = [0u8; 2];
    for i in 0..64 {
        let iu16 = i as u16;
        handle.read_control(libusb::request_type(Direction::In, RequestType::Vendor, Recipient::Interface), 0x90, 0, iu16 << 8, &mut buf, Duration::from_secs(10)).unwrap();
        assert_eq!(buf[0], ROM[2*i]);
        assert_eq!(buf[1], ROM[2*i + 1]);
        println!("{}", i);
    }
}
```

For some reason, one of the control_ins will time out at a random point in the loop:
```
0
1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Timeout', src/main.rs:21:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

What I know so far:
* The test program runs on occasion, but usually times out
* The closed-source binary equivalent handles these repeated control_in reads correctly every time. It is based off of the Arduino-equivalent [ArduinoCore-samd](https://github.com/arduino/ArduinoCore-samd/tree/master/cores/arduino/USB) USB library
* The timeout does not put the device in an unusable state; running the test program again works just fine until another timeout/occasional success

I would really appreciate your help in root-sourcing the cause of this timeout with usb-device. I'm not sure how to use the UART or RRT debugging in atsamd-hal, so haven't been able to get any logs of what the device is doing when this is happening.

mvirkkunen, 2020-03-27:
Just to make sure, are you running in release mode? Debug mode code in Rust is sometimes extremely slow and could fail to meet USB timing requirements.

sameer, 2020-03-27:
Yes, I am running in release mode. I use objcopy to generate a binary, then bossac to flash it. I did set optimize for size, lto = true and codegen units = 1 so I'll try disabling those.

Thank you for getting back to me so quickly.

-------- Original Message --------

mvirkkunen, 2020-03-27:
If you replace the loop in "fn main" with just an empty "loop { }", does anything change? (I assume the control transfers should still work)

sameer, 2020-03-27:
> If you replace the loop in "fn main" with just an empty `loop { }`, does anything change? (I assume the control transfers should still work)

I think I just figured it out. I removed the code in the loop & the poll, and it started working more often. Turns out it was all because I set the optimization profile to size...which made it slow enough to work only sometimes.

Interestingly, Arduino gcc optimizes for size too so I didn't think that would be the cause.

mvirkkunen, 2020-03-27:
Code that's optimized for size shouldn't be *that* slow, debug mode is usually the only option that causes things to actually fail. My suspicion was that because your main loop held essentially a permanent critical section that the interrupts weren't being served correctly/quickly enough. Currently you're essentially not using interrupts at all and may as well put the "USB" interrupt handler code in the main loop.

Disasm, 2020-03-27:
Maybe the core frequency is not big enough, it can also affect the performance.

sameer, 2020-03-27:
> Code that's optimized for size shouldn't be _that_ slow, debug mode is usually the only option that causes things to actually fail. My suspicion was that because your main loop held essentially a [...]

When I put the interrupt code in the main loop, the device still shows up but my test program times out.

sameer, 2020-03-29:
Ok, I got the test to work successfully every time. Here's what I changed / found out :

* The `arrayvec` crate I was using for statically-sized buffers wasn't working at all, so I just directly used arrays and slices (probably the problem)
* The board I'm using has an external oscillator so I switched my clock setup to use that
* For bulk endpoints, you **MUST** allocate both an in and an out endpoint or reading/writing does not work at all. If you want an endpoint that is output only, allocate both the in/out endpoints but only include the out endpoint in the configuration descriptor.

closed 2020-03-29 by sameer

mvirkkunen, 2020-03-29:
To me the oscillator reason sounds most plausible. The internal oscillator could be either too slow or too imprecise. Basic internal RC oscillators don't usually fill the precision requirements for USB, so you need either an external crystal or clock recovery to tune the internal oscillator (not sure if your chip has that).

The other reasons sound pretty odd. `arrayvec` is fairly popular and should work if used correctly, and there's no reason for bulk transfers to require an endpoint in both directions. The latter could be a bug in the USB driver for your specific chip - if you have the time it would be nice if you could test to make sure that is definitely broken and report an issue on the driver.

However before doing that you should preferably get some kind of debug logging working so you can make sure your loops are actually running and most importantly that your code doesn't panic without you knowing

sameer, 2020-03-30:
> To me the oscillator reason sounds most plausible. The internal oscillator could be either too slow or too imprecise. Basic internal RC oscillators don't usually fill the precision requirements for [...]

There are calibration parameters in the non-volatile memory for the microcontroller used to trim the USB lines. I don't know about the clock recovery though.

> The other reasons sound pretty odd. `arrayvec` is fairly popular and should work if used correctly, 

The crate itself is fine and I have used it before, just that it didn't make sense to use here. Dereferencing an arrayvec gives you a slice of the current contents. So essentially I was asking to read the out endpoint with an empty mutable slice because I was thinking of it like [Read::read_to_end](https://doc.rust-lang.org/std/io/trait.Read.html#method.read_to_end). Just using a fixed size array made more sense here.

> and there's no reason for bulk transfers to require an endpoint in both directions. The latter could be a bug in the USB driver for your specific chip - if you have the time it would be nice if you could test to make sure that is definitely broken and report an issue on the driver.

I'll investigate this.

> However before doing that you should preferably get some kind of debug logging working so you can make sure your loops are actually running and most importantly that your code doesn't panic without you knowing

I've been using the builtin LED light on my board as a sort of canary, since it won't swap to open drain and light up if there's a panic. How do you interface with the uart dbgprint? Do I need some sort of USB to UART board?

mvirkkunen, 2020-03-30:
@sameer There are many ways to debug ARM boards and you should look into what's available on your specific one. You can use UART, ITM, semihosting, RTT, et cetera. Mostly everything besides UART requires a debug probe that interfaces with the chip using SWD. For UART debugging a USB serial converter will work, but I'm not sure if there's a crate that outputs panic messages over a UART.

sameer, 2020-04-02:
Ok, I took some time and checked the things I changed. With the internal oscillator and without the extra endpoints I created, it is working fine.

I looked through my commit history and seems like the pipe errors were just because I was trying to use an arrayvec as a slice buffer. The built-in LED only told me if things initialized successfully, not if there was a runtime panic, so I had no way of knowing that that was the problem.

Thank you for following up on this Matti, and sorry for all the confusion.