rust-embedded-community/usb-device #47: USB read doesn't trigger

dimpolo, 2020-10-19:
I am stuck trying to implement a simple interface using this driver.
My knowledge of the USB stack is very limited so I'm not sure were to look or even in which crate to ask, so if this issue belongs somewhere else please let me know.

I am stuck with some strange behavior where data from the host seemingly reaches my board but the `poll` method returns `false` until I send something to the host.
Also telling the host to send the data slowly or with excess data seems to solve the problem.

I'm not sure where I might be doing something wrong so I'll try and post most of the code. Sorry for the wall of text.
Thanks for helping out.

**STM32F405RG with known good USB connection**
```rust
#![no_main]
#![no_std]

use cortex_m_rt::entry;
use panic_halt as _;
use stm32f4xx_hal::otg_fs::{UsbBus, USB};
use stm32f4xx_hal::{otg_fs, prelude::*, stm32};
use usb_device::device;
use usb_device::prelude::*;
use usbd_serial::SerialPort;

type Serial<'a> = SerialPort<'a, otg_fs::UsbBus<otg_fs::USB>>;
type UsbDevice<'a> = device::UsbDevice<'a, otg_fs::UsbBus<otg_fs::USB>>;

static mut EP_MEMORY: [u32; 1024] = [0; 1024];

const COMMAND_BYTE: u8 = 8;

pub struct UsbSerial<'a> {
    usb_device: UsbDevice<'a>,
    serial: Serial<'a>,
}
impl<'a> UsbSerial<'a> {
    pub fn new(usb_device: UsbDevice<'a>, serial: Serial<'a>) -> Self {
        UsbSerial { usb_device, serial }
    }

    fn poll(&mut self) -> bool {
        self.usb_device.poll(&mut [&mut self.serial])
    }

    fn write_all(&mut self, buf: &[u8]) {
        let count = buf.len();

        let mut write_offset = 0;
        while write_offset < count {
            if let Ok(len) = self.serial.write(&buf[write_offset..count]) {
                write_offset += len;
            }
        }
    }

    fn recv(&mut self, buf: &mut [u8]) -> Option<usize> {
        if self.poll() {
            if let Ok(count) = self.serial.read(buf) {
                if count > 0 {
                    return Some(count);
                };
            }
        }
        None
    }

    fn read_all(&mut self, buf: &mut [u8]) {
        let count = buf.len();
        let mut read_offset = 0;
        while read_offset < count {
            if let Some(len) = self.recv(&mut buf[read_offset..count]) {
                read_offset += len;
            }
        }
    }
}

#[entry]
fn main() -> ! {
    let dp = stm32::Peripherals::take().unwrap();

    let rcc = dp.RCC.constrain();
    rcc.cfgr.use_hse(12.mhz()).sysclk(168.mhz()).freeze();

    let gpioa = dp.GPIOA.split();

    let usb = USB {
        usb_global: dp.OTG_FS_GLOBAL,
        usb_device: dp.OTG_FS_DEVICE,
        usb_pwrclk: dp.OTG_FS_PWRCLK,
        pin_dm: gpioa.pa11.into_alternate_af10(),
        pin_dp: gpioa.pa12.into_alternate_af10(),
    };

    let usb_bus = UsbBus::new(usb, unsafe { &mut EP_MEMORY });

    let serial = usbd_serial::SerialPort::new(&usb_bus);

    let usb_dev = UsbDeviceBuilder::new(&usb_bus, UsbVidPid(0x16c0, 0x27dd))
        .manufacturer("Fake company")
        .product("Serial port")
        .serial_number("TEST")
        .device_class(usbd_serial::USB_CLASS_CDC)
        .build();

    let mut usb_serial = UsbSerial::new(usb_dev, serial);

    loop {
        let mut command_byte = [0];
        usb_serial.recv(&mut command_byte);
        if command_byte[0] == COMMAND_BYTE {
            // usb_serial.write_all(b"fix ");  everything works fine if I add this line

            let mut buf = [0; 4];
            usb_serial.read_all(&mut buf);  // try to receive 4 bytes
            usb_serial.write_all(b"got it");  // never get here (except with the workarounds)
        }
    }
}
 ```

**Host Side:**
``` python
import serial
import time

COM_PORT = 'COM13'

com_port = serial.Serial(timeout=0.5, port=COM_PORT)

COMMAND_BYTE = b'\x08'
EXTRA_BYTE = b'X'
BYTES = b'1234'

def works():
    for byte in COMMAND_BYTE + BYTES:
        com_port.write(bytes([byte]))
        time.sleep(0.1)
    print(com_port.read_all())

def sometimes_works():
    com_port.write(COMMAND_BYTE + BYTES)
    time.sleep(0.5)
    com_port.write(EXTRA_BYTE)
    print(com_port.read_all())

def does_not_work():
    com_port.write(COMMAND_BYTE + BYTES)
    time.sleep(0.1)
    print(com_port.read_all())
```

Disasm, 2020-10-20:
`poll` will not return true if the incoming packet request is already processed or skipped on the previous iteration. You should `read` the `serial` until it stops returning the data. `SerialPort` buffers the incoming data, so `poll` returning false doesn't mean you have no data to read.

dimpolo, 2020-10-20:
Thanks a lot!
That cleared things up.

I edited the question and removed irrelevant bits, so it can be more helpful to others.