rust-embedded-community/usb-device #48: UsbDevice lifetime preventing accessing from interrupt

rukai, 2020-11-23:
To correctly use UsbDevice we should hook `poll` up to an interrupt to call it regularly.

However, as far as I can see that is impossible with the current API design.
Due to lifetimes `UsbDevice` needs to outlive the `UsbBusAllocator` passed to `UsbDeviceBuilder`.
Because of that we cant put it in a global storage like:
`static USB_DEVICE: Mutex<RefCell<Option<UsbDevice<UsbBus<USB>>>>>;`
And without global storage there is no way to call `poll` in the interrupt
To resolve this `UsbDeviceBuilder::new` should take a `&UsbBusAllocator`, `UsbBusAllocator`, or `Arc<UsbBusAllocator>` instead of `&'a UsbBusAllocator`.

Now it is quite possible that my reasoning above is wrong somewhere, i am no master of lifetimes.
And it does seem strange to me that something like this would be overlooked...

So if there is a way, please add an example to https://github.com/stm32-rs/stm32-usbd-examples that demonstrates calling `poll` from an interrupt. Or if you give me enough information I might be able to implement such an example myself.

mvirkkunen, 2020-11-23:
For interrupts you'll need a `&'static UsbBusAllocator`.

Here's one example that does it: https://github.com/stm32-rs/stm32-usbd-examples/blob/master/example-stm32f103c8/examples/serial_interrupt.rs

rukai, 2020-11-23:
Ah, thankyou!
I had incorrectly assumed that each `example-sfm32f*` folder contained the same single example but ported for that board.

Would you accept a PR to add a list of each example to the readme so make it easier to look through the examples for every board at once?

Disasm, 2020-11-23:
Examples now live in the corresponding hal repositories, so this stm32-usbd-example repo is kinda outdated.

rukai, 2020-11-23:
I see.
Maybe we should change the link in https://github.com/mvirkkunen/usb-device/blob/master/README.md#hardware-driver-crates to list examples in the various hals such as https://github.com/stm32-rs/stm32f1xx-hal/blob/master/examples/usb_serial_interrupt.rs instead?

mvirkkunen, 2021-03-13:
I decided to remove the link altogether and just suggested people look in the individual HALs.

closed 2021-03-13 by mvirkkunen