rust-embedded-community/usb-device #52: MIT license compliance w.r.t. embedded applications

agausmann, 2021-01-29:
I would love to use this in my embedded projects, but I'm not comfortable using MIT-licensed code in such applications, because of the condition:

> The above copyright notice and this permission notice shall be included in all
copies or substantial portions of the Software.

Granted, it does just say "included", not "displayed", so I guess I could include the license somewhere in the firmware binary and it would be okay? But I feel like being able to view the license is somewhat implied, so I'm not satisfied with this.

In many embedded applications, there's no easy way to prominently "display" license notices like this, and I think the (implied) obligation to do so is an unnecessary burden. For example, if I'm writing firmware for a simple keyboard or mouse, how would I display the license notice from the device?

So, I think MIT is not a great fit for libraries used in the embedded ecosystem. Alternatively, I would suggest the Apache-2.0 license, as that is the one most used by the Rust Embedded WG, and only requires a license notice for copies of the source code form. Also, to not "break compatibility" with the current licensees, I would suggest retaining the MIT license, and dual-licensing with Apache-2.0.

agausmann, 2021-01-29:
This also applies to your device class crates like `usbd-hid`.

wez, 2021-02-04:
[IANAL](https://en.wikipedia.org/wiki/IANAL), but: for MIT style licenses, it's usually enough to include the license in supporting documentation (which these days can be eg: a page on your website somewhere) for the case where you're shipping the compiled version.

Note that the Apache 2.0 license includes a grant of patent license with a revocation clause which is a bit more fraught with legal and licensing complexity.

MIT is about as permissive a license as you could wish for!

Note that you should speak with an appropriately qualified lawyer if you are looking to protect yourself and your projects, rather than trusting "random people" on the internet :)

mvirkkunen, 2021-03-13:
Just out of interest, how would one even go about adding a new license to a project? Does it require approval from *every* contributor ever?

wez, 2021-03-13:
Regarding re-licensing, in general, yes, you'll need to get every contributor to agree to have their contribution distributed under the proposed license terms, or you'll need to back out their contribution and re-implement it.  Some projects employ a CLA that grants the project the right to make future licensing adjustments without further consultation with the contributor.

leo60228, 2021-05-11:
I would personally find it surprising if that condition couldn't be met by including the notice to the documentation.

Notably, a large portion of the Rust ecosystem (including the standard library) is dual-licensed under MIT and Apache 2.0. Apache 2.0 is much more specific on this subject:

> You must give any other recipients of the Work or Derivative Works a copy of this License; and
> You must cause any modified files to carry prominent notices stating that You changed the files; and
> You must retain, in the Source form of any Derivative Works that You distribute, all copyright, patent, trademark, and attribution notices from the Source form of the Work, excluding those notices that do not pertain to any part of the Derivative Works; and
> If the Work includes a "NOTICE" text file as part of its distribution, then any Derivative Works that You distribute must include a readable copy of the attribution notices contained within such NOTICE file, excluding those notices that do not pertain to any part of the Derivative Works, in at least one of the following places: within a NOTICE text file distributed as part of the Derivative Works; within the Source form or documentation, if provided along with the Derivative Works; or, within a display generated by the Derivative Works, if and wherever such third-party notices normally appear. The contents of the NOTICE file are for informational purposes only and do not modify the License. You may add Your own attribution notices within Derivative Works that You distribute, alongside or as an addendum to the NOTICE text from the Work, provided that such additional attribution notices cannot be construed as modifying the License.
> 
> You may add Your own copyright statement to Your modifications and may provide additional or different license terms and conditions for use, reproduction, or distribution of Your modifications, or for any such Derivative Works as a whole, provided Your use, reproduction, and distribution of the Work otherwise complies with the conditions stated in this License.

ryan-summers, 2022-05-23:
Foreword: I am also not a layer ;)

There are plenty of discussions on MIT licenses in embedded applications, e.g.:
* https://opensource.stackexchange.com/questions/8365/mit-vs-gnu-licences-for-embedded-system-projects
* https://www.quora.com/Can-I-embed-a-software-with-an-MIT-license-inside-my-application
* https://www.tawesoft.co.uk/kb/article/mit-license-faq

Many of the above links indicate that it is likely sufficient to include the license in external documentation (or via malicious compliance where the license text is in the binary application - MIT does not require that the license is easily readable). There are many approaches to satisfying the MIT license.

However, MIT is a commonly used license within Rust, and this software has been distributed under MIT licensing for quite a while. While we could change licensing over to one that does not require notice (e.g. 0BSD), we would require approval of all previous contributors.

Given that Github is intended as a developer platform and we're not able to provide legal advice, I'm going to close this issue. If you would like to propose alternative licensing and have approval from contributors, we can handle that separately.

closed 2022-05-23 by ryan-summers