rust-embedded-community/usb-device #72: CTR_RX is not being cleared.

linasdev, 2021-06-07:
When the direction is `IN` and `ep_setup` and `ep_out` bits are false, the `CTR_RX` register is not being cleared, so my USB polling interrupt keeps firing and the program stalls. This might be completely an issue on my end but I would still appreciate some pointers in the general direction. As you can tell, I have little USB experience. I'm just making a tiny project and thought Rust would be a nice touch. I'm using `usbd-hid`, `stm32-usbd` (I'm running this on a stm32f103cbt6) and, obviously, `usb-device` itself.

https://github.com/mvirkkunen/usb-device/blob/b6037b0f98c933ff0a18a864fd34a42f89a6316a/src/device.rs#L206-L212

mvirkkunen, 2021-06-07:
When the direction is `IN`, there should be no receiving going on as `IN` means device to host direction (TX not RX from the device's viewpoint).

In `stm32-usbd`, reading the packet from the endpoint clears the interrupt flag. Are you reading from the endpoints you're using (e.g. `pull_raw_output` in `usbd-hid`)?

linasdev, 2021-06-07:
Hey, so, yeah, that was exactly it. I'm sorry for wasting your time. Closing the issue.

closed 2021-06-07 by linasdev

stappersg, 2021-06-08:
On Mon, Jun 07, 2021 at 03:29:02PM -0700, Linas Nikiperavičius wrote:
> Hey, so, yeah, that was exactly it. I'm sorry for wasting your time. Closing the issue.
> 

Next time:

Hey, so, yeah, that was exactly it. Thanks for making it clear.
I leave the issue open for a fews days, so more people gain from it.