rust-embedded-community/usb-device #80: The USB set address request failed.

zgdump, 2022-01-01:
Hi!

I used your library in my project for STM32F411CEU and noticed that sometimes Windows shows error 43 in Device Manager. If reconnect device, then everything works as it should

To reproduce the issue, I plugged and popped the device several times. Sometimes it also occurred if leave the stm32 connected via USB and flash using st-link

[image]

zgdump, 2022-01-01:
I am using a Chinese black pill. And everything works fine on firmware, written using C and CubeMX 

[image]

zgdump, 2022-01-01:
Hypothetically, i can try to get the dump using a cheap logic analyzer, but I'm not sure that it will be able to work correctly with usb 2.0

zgdump, 2022-01-01:
I can't figure out what is what and whether my logical analyzer managed to capture data from usb

[RUST USB CDC BUG.zip](https://github.com/mvirkkunen/usb-device/files/7798355/RUST.USB.CDC.BUG.zip)

mvirkkunen, 2022-01-02:
Can you paste your list of dependencies from `Cargo.toml` with the versions numbers?

zgdump, 2022-01-02:
My Cargo.toml:

```toml
[package]
name = "motor"
version = "3.0.0"
edition = "2021"

[dependencies]
nb = "1"
cortex-m = "*"
cortex-m-rt = "*"
embedded-hal = "0.2"
usb-device = "0.2.8"
usbd-serial = "0.1.1"
rtt-target = { version = "0.3.1", features = ["cortex-m"] }
arraydeque = { version = "0.4", default-features = false }

[dependencies.stm32f4xx-hal]
version = "0.10"
features = ["stm32f411", "rt", "usb_fs"]
```

zgdump, 2022-01-02:
@mvirkkunen 

I'm sorry, I made a mistake. Before that, I started the project in debug mode using the `cargo-embed --chip stm32f411ceu` command. After I started launching the project via `cargo-embed --chip stm32f411ceu --release`, the problem went away even at low frequencies. But this is still wrong behavior, I think

mvirkkunen, 2022-01-02:
Yes, it's odd. It shouldn't be *that* slow and the dump shows a STALL condition, not a timeout from what I could tell from a quick look. I'll have to take a look at this more when I have time, I have some of the exact same board too.

zgdump, 2022-01-02:
If this problem is solved one day, I will be glad. Thank you for communicating! For now, I will work in the release and use your rtt library for debugging

My friend sent me such a link when I described my problem to him: https://discord.com/channels/705836771304538152/856573424075538503/927284955057373234

mvirkkunen, 2022-01-02:
Yeah there are some weird race conditions in debug mode still it seems. What comes to debugging, stopping the device or even using semihosting is just a plain non-starter because USB has somewhat strict timeouts. I pretty much made the first version of the RTT library specifically for debugging USB, because semihosting was too slow and I didn't want to use an extra serial port.

I can't see what's behind the link, because I don't know what Discord server it is and I don't think it allows you to see messages unless I join the server.

zgdump, 2022-01-02:
> Yeah there are some weird race conditions in debug mode still it seems. What comes to debugging, stopping the device or even using semihosting is just a plain non-starter because USB has somewhat [...]

Excuse me. Corrected the link.It's time for me to sleep)

mvirkkunen, 2022-01-02:
> The Synopsys OTG IP is very fragile and easily breaks down in surprising ways when it's not handled exactly as the IP authors intended, even if you'd think your actions are reasonable.

mmm, sounds very nice indeed

zgdump, 2022-01-02:
> mmm, sounds very nice indeed

Another surprise from stm32) It's amazing why nobody didn't encounter this problem besides me

Disasm, 2022-01-02:
It's quite an old problem, there is even a warning which (unfortunately) doesn't work unless you build the firmware from a workspace which includes synopsys-usb-otg: https://github.com/stm32-rs/synopsys-usb-otg/blob/master/build.rs#L5-L7

mvirkkunen, 2022-01-02:
It's possible people have just assumed it was because debug mode is slow and can break in weird ways, and never looked at it with a logic analyzer to see the surprise STALL status. This report might be a better fit for the Synopsys driver repo, unfortunately you can't move an issue to another org.

zgdump, 2022-01-02:
> It's possible people have just assumed it was because debug mode is slow and can break in weird ways, and never looked at it with a logic analyzer to see the surprise STALL status. This report might be a better fit for the Synopsys driver repo, unfortunately you can't move an issue to another org.

I can write anywhere, summarizing everything written, if it helps a little in solving the problem.

zgdump, 2022-01-02:
Should I drop it here? https://github.com/stm32-rs/synopsys-usb-otg
Or where it will be more convenient

Disasm, 2022-01-02:
Yes, this is the right place.

zgdump, 2022-01-02:
Close issue here...

closed 2022-01-02 by zgdump

zgdump, 2022-01-02:
And open here https://github.com/mvirkkunen/usb-device/issues/80