rust-embedded-community/usb-device #89: embassy-usb

stappersg, 2022-05-19:
In #79 did I learn there is [`embassy-usb`](https://github.com/embassy-rs/embassy/blob/master/embassy-usb/src/driver.rs).
With this issue I want to share that information with you. (With this issue am I sharing that information with you.)

ryan-summers, 2022-05-21:
Closing this as not-an-issue. We can point people at embassy-usb via the README, but tracking it via the issue tracker doesn't seem like the right approach here, as there's nothing we can immediately act on. 

If you'd like to improve discoverability of embassy-usb, we can happily accept a PR for the README :)

closed 2022-05-21 by ryan-summers

stappersg, 2022-05-21:
On Sat, May 21, 2022 at 04:28:48AM -0700, Ryan Summers wrote:
> Closing this as not-an-issue. We can point people at embassy-usb via
> the README, but tracking it via the issue tracker doesn't seem like
> the right approach here, as there's nothing we can [...]

That became https://github.com/rust-embedded-community/usb-device/pull/92

Groeten
Geert Stappers
-- 
Silence is hard to parse