rust-embedded-community/usb-device #102: Can't implement CDC with HID

yallxe, 2022-08-07:
I created a CDC and HID classes with one device
```rust
let mut hid = HIDClass::new_with_settings(
    &bus_allocator,
    KeyboardReport::desc(),
    10,
    HidClassSettings {  
        subclass: HidSubClass::NoSubClass,
        protocol: HidProtocol::Keyboard,
        config: ProtocolModeConfig::ForceReport,
        locale: HidCountryCode::NotSupported
    }
);
let mut serial = SerialPort::new(&bus_allocator);

let mut dev = UsbDeviceBuilder::new(&bus_allocator, vid_pid)
    .manufacturer("xkbd")
    .product("Prism80")
    .serial_number("000")
    .device_class(USB_CLASS_CDC)
    .build();
```

And the loop code:
```rust
loop {
    dev.poll(&mut [&mut serial, &mut hid]);
    if button_pin.is_high().unwrap() {
        hid.push_input(&report_struct).ok();
        led_pin.set_high().ok();
        serial.write(b"Pressed").ok();
    } else {
        hid.push_input(&null_report_struct).ok();
        led_pin.set_low().ok();
        serial.write(b"Released").ok();
    }

    // drop received data
    hid.pull_raw_output(&mut [0; 64]).ok();
}
```
However the USB device is not recognized by my PC, I tried the solution from #85 but it does not fix the issue and the device keep being unrecognized, how do I implement the HID with CDC?

Disasm, 2022-08-07:
Do you have a `dmesg` log of failed enumeration? Are you using a release build?

yallxe, 2022-08-07:
I don't, I'm kind of new to embedded developement and as far as I understand I need to have a debug probe to get this, maybe I understand it wrong. 

P.S. I'm using Pi Pico.

Disasm, 2022-08-07:
To get `dmesg` logs you need Linux (MacOS probably supports these as well).
By the way, does it work for you if you add only HID or only CDC?

yallxe, 2022-08-07:
I'm on Windows currently, I can try WSL or use my laptop on linux and yes it does work with CDC or HID only.

yallxe, 2022-08-07:
Just tested on my laptop, and... HID works, but I can't connect to serial port. I tried to connect using `screen /dev/ttyACM1 115200` but I'm getting a `[screen is terminating]` error. However on Windows, things seems to be even worser, I don't really know how to check for errors and the device keep being unrecognized.

Also checked `dmesg` and usb device seems to be created with no errors.

yallxe, 2022-08-07:
Forgot to answer, yes I'm using release build.

Disasm, 2022-08-07:
Hmm, could you check if it works in Windows without `.device_class(USB_CLASS_CDC)`? Also it would be interesting to see USB capture (for example, gathered with Wireshark) which contains packets transferred when you connect to the device with `screen`.

ianrrees, 2022-08-07:
I think one issue may be that you're calling `serial.write()` every time through the `loop {}`.  In practice, doing one serial write will need >1 pass through the loop to finish, so I think eventually you'll have a problem with that approach.  Maybe only send the message when the button level changes - something like this:

```Rust
let mut last_state = button_pin.is_high().unwrap();
loop {
    // Poll stuff as above.  I'm not sure how HID works so maybe keep it as-is, maybe adapt as below

    let current_state = button_pin.is_high().unwrap();
    if last_state != current_state {
        last_state = current_state;
        if current_state {
            serial.write(b"Pressed").ok();
        } else {
            serial.write(b"Released").ok();
        }
}
```

ianrrees, 2022-08-07:
I'd also leave out the line `.device_class(USB_CLASS_CDC)` - device class should probably be 0 (the default) which tells the host OS to figure out the class of each interface from its descriptors.

yallxe, 2022-08-08:
I know about calling `serial.write()` every loop, fixed that already but didn't mention here, also removed `.device_class(USB_CLASS_CDC)` but it doesn't fix the problem.

yallxe, 2022-08-08:
Okay, I managed to open serial monitor with `screen`, I was getting `[screen is terminating]` because I forgot to run as sudo. So basically, everything works fine on Linux, but not on Windows.

ryan-summers, 2022-08-08:
What version of `usb-device` are you using? There was an enumeration bug on Windows that was released in 2.9: https://github.com/rust-embedded-community/usb-device/blob/master/CHANGELOG.md

If you don't have that fix implemented, it's highly unlikely that your device will enumerate on Windows.

mvirkkunen, 2022-08-08:
Have you tried adding interface association descriptors (IADs)? Windows is usually more picky about them than other OSs.

You can add them by removing `device_class` / `device_subclass` / `device_protocol` and just calling `composite_with_iads()` instead of them.

yallxe, 2022-08-08:
Just updated `usb-device` to `0.2.9` and added `composite_with_iads()`, but still no results. Also made sure I'm using newest `usbd-serial 0.1.1` and `usbd-hid 0.6`. Maybe something is wrong in my general code, so I'm also uploading it [here](https://gist.github.com/yallxe/4201885c8debd335d08a085849a9bd99).

stappersg, 2022-08-08:
This is a comment that will soon be deleted.  Do know that I care about embedded rust and [don't want to feed github](https://sfconservancy.org/GiveUpGitHub/).

> Just updated usb-device to 0.2.9 and added composite_with_iads(), but still no results. Also made sure I'm using newest usbd-serial 0.1.1 and usbd-hid 0.6. Maybe something is wrong in my general code, so I'm also uploading it [here](https://gist.github.com/yallxe/4201885c8debd335d08a085849a9bd99).

Put it in a git repository,  so it can be `git clone`d and `cargo build`.  It will also help explaining that usb-device version 0.2.9  is a rust crate or a library for some MS Windows version.

yallxe, 2022-08-12:
Changing `max_packet_size_0` to `16` instead of default `8` seems to fix the problem. Would be glad if somebody could explain me why was this occurring only on Windows, but as a problem is resolved, I'm closing the issue. Thanks everybody for the help.

closed 2022-08-12 by yallxe