rust-embedded-community/usb-device #109: unresolved import `core::sync::atomic::AtomicPtr`

alexmadeathing, 2022-12-06:
I'm getting the following error when compiling for `atmega32u4`. Any ideas?
```
error[E0432]: unresolved import `core::sync::atomic::AtomicPtr`
 --> C:\Users\user\.cargo\registry\src\github.com-1ecc6299db9ec823\usb-device-0.2.9\src\bus.rs:6:26
  |
6 | use core::sync::atomic::{AtomicPtr, Ordering};
  |                          ^^^^^^^^^
  |                          |
  |                          no `AtomicPtr` in `sync::atomic`
  |                          help: a similar name exists in the module: `AtomicI8`
```

mvirkkunen, 2022-12-06:
That would be because AVR doesn't have atomic types (of that size), so the standard library does not implement them.

Using [atomic-polyfill](https://crates.io/crates/atomic-polyfill) would be an option these days, without changing the rest of the architecture.

alexmadeathing, 2022-12-06:
Ahhh right. I presume that would just be a matter of forking usb-device and replacing `core::sync::atomic::AtomicPtr` with `atomic_polyfill::AtomicPtr`?

I'll try it out later today.

I wonder then if it would be worth providing a feature in usb-device, which does this.

alexmadeathing, 2022-12-06:
I've also seen this, which mentions atomics on AVR: https://crates.io/crates/portable-atomic

mvirkkunen, 2022-12-06:
That crate looks like it does a similar thing. As far as I know both of these are supposed to be drop-in replacements for platforms with atomics, and only require extra configuration on platforms that do not support atomics, so a feature might not even be necessary.

alexmadeathing, 2022-12-06:
As I was implementing it, I noticed the same thing and was about to type exactly what you said. I might format this as a PR once I've confirmed it working.

alexmadeathing, 2022-12-06:
Ok that's compiled. In order to test I'm going to have to finish writing my usb device test code. Hopefully shouldn't take too long.

My usb-device branch with polyfill alteration can be found here: https://github.com/alexmadeathing/usb-device/tree/atomic-polyfill

alexmadeathing, 2022-12-06:
It seems I should also be able to set the `max-atomic-width` in my target spec.

https://github.com/Rahix/avr-hal/issues/40#issuecomment-740626847

alexmadeathing, 2022-12-07:
Ok that's what I'm going with. I'll be deleting my fork and I'll close this issue.

closed 2022-12-07 by alexmadeathing

mvirkkunen, 2022-12-07:
@alexmadeathing Isn't that for native atomics? Does AVR actually have atomic instructions? Or does it pull in the clang builtins that the rest of Rust seems to avoid?

alexmadeathing, 2022-12-07:
I don't know tbh. I guessed that if it compiled, there's probably a working intrinsic implementation. The atmega32u4 datasheet has example code for how to implement them and I've seen chat on the Rust project mentioning avr atomics, but I couldn't find anything definitive that suggests how they've implemented them. I'm slightly out of my depth though as this is my first rust embedded project.

From the datasheet section 14.2, it seems there are no atomics at the instruction level, but they provide this example implementation:
```asm
TIM16_ReadTCNTn:
; Save global interrupt flag
in r18,SREG
; Disable interrupts
cli
; Read TCNTn into r17:r16
in r16,TCNTnL
in r17,TCNTnH
; Restore global interrupt flag
out SREG,r18
ret
```