rust-embedded-community/usb-device #125: Changelog: add another `EndpointType` changelog to "Breaking"

eZioPan, 2023-07-29:
https://github.com/rust-embedded-community/usb-device/blob/7ebb10fe5f952955b416c5b05abc84d37bf82acf/CHANGELOG.md?plain=1#L22-L23

Since `EndpointType` is a public enum, and some crates [may use the integer form of it](https://github.com/stm32-rs/synopsys-usb-otg/blob/aa12f1a4e21441159a1f3a646768f43cbb3e74b9/src/endpoint.rs#L92-L99). After this modification user should call `.to_bm_attributes()` to access numeric form. We might add a another changelog to tell user about this.

```diff
### Breaking
+++ * Acess numeric form of `EndpointType` variants now require a `.to_bm_attributes()` ([#60](https://github.com/rust-embedded-community/usb-device/pull/60)).
```

eldruin, 2023-08-02:
Thanks for noticing this.
Could you put this into a PR?

eZioPan, 2023-08-02:
please review my PR

PR #126 (2023-08-02): Update CHANGELOG.md
will close #125
closed 2023-08-02 by eldruin