rust-embedded-community/usb-device #135: Feature request: TinyUSB integration

GamePad64, 2023-11-19:
Hi! I am developing apps for ESP32-S2/S3 platform. These chips are not supported by usb-device, but they are compatible with tinyUSB stack. It will be great to create a thin wrapper, that implements UsbBus trait and uses tinyUSB as backend.  It will provide a workaround for a lot of devices, until there will be a native driver.

ryan-summers, 2023-11-19:
So usb-device _is_ the USB device-side stack (along with the class drivers), so it's a replacement for tinyUSB on embedded devices. As such, I'm not entirely sure how it would make sense to add a shim into the one usb device stack for another, as they're just two different avenues to do the same thing.

ryan-summers, 2024-02-15:
Closing, as I do not think there's further activities here in this crate.

closed 2024-02-15 by ryan-summers