rust-embedded-community/usb-device #139: Improvement: Minimize LANGID enum under feature flag

maor1993, 2024-01-19:
Recently I started mitigating my code to use usb-device 0.3.1 on an stm32l4 project.

looking the objdump I noticed the langid conversion method requires ~8KB of code:
[image]
clearly, mapping a huge enum this is unavoidable.
however,
to support smaller builds I think hiding most langIDs under a feature flag would suffice.

as a test run I copied the langid enum and covered it with a feature flag `min-lang-id`

```rust
#[allow(missing_docs)]
#[derive(Clone, Copy, PartialEq, Debug, TryFromPrimitive)]
#[cfg_attr(feature = "defmt", derive(defmt::Format))]
#[repr(u16)]
#[cfg(feature="min-lang-id")]
pub enum LangID {
    EN_US = 0x0409,
}
```

assuming we minimize to only EN_US, the enum gets completly optimized out.
I'm sure there can be a minimal set of common LANGIDs actually seen in the wild.
Regardless, I do thing that avoid adding this much code for an embedded targeted code is essential.

ryan-summers, 2024-01-19:
I would not be opposed to this idea. That's wild that it takes so much space though.

vitalyvb, 2024-01-28:
It's a very noticeable size increase if usb-device is used in a bootloader.

I can see why this API is better than a plain u16, however it's not obvious why usb-device is supposed to validate and check language id and not leave it to a user of library.

Maybe something like this?
```rust
#[allow(missing_docs)]
#[derive(Clone, Copy, PartialEq, Debug)]
pub struct LangID(u16);

impl From<LangID> for u16 {
    fn from(lang_id: LangID) -> Self {
        let LangID(id) = lang_id;
        id
    }
}

impl TryFrom<u16> for LangID {
    type Error = &'static str;

    fn try_from(lang_id: u16) -> Result<Self, Self::Error> {
        Ok(LangID(lang_id))
    }
}

#[allow(missing_docs)]
impl LangID {
    pub const AR: LangID = LangID(0x0001);
    pub const BG: LangID = LangID(0x0002);
    pub const CA: LangID = LangID(0x0003);
    pub const ZH_HANS: LangID = LangID(0x0004);
    pub const CS: LangID = LangID(0x0005);
    pub const DA: LangID = LangID(0x0006);
    pub const DE: LangID = LangID(0x0007);
    pub const EL: LangID = LangID(0x0008);
    pub const EN: LangID = LangID(0x0009);
...
```

ryan-summers, 2024-01-29:
I like that idea of just having constants instead if it eliminates the code bloat. LangIDs aren't even part of the USB spec anymore.

dalegaard, 2024-02-15:
This was taking up a big chunk of memory on my current target, so I had to change this on my local fork. To keep the safety in the developer API, I made `StringDescriptors` keep the LangID parameter, but removed the `TryFromPrimitive` and let the internal parts of `usb_device` simply handle the u16. For my use case, there's no need to actually validate the received language is valid, so working simply in the u16 saves a ton of memory(was around 3kB).

Best regards,
Lasse

ryan-summers, 2024-02-15:
If someone submits a PR, I'd be happy to review, merge, and release it :)

The whole concept of LangIDs is a bit questionable in my opinion since I think they're deprecated now as well, so simplifying this sounds like a wonderful idea.

PR #141 (2024-02-28): Convert LangID from u16 enum to tuple struct containing u16
Relax lang_id checks provided by enum so that any lang_id could be used, not only values listed in the enum. These checks increased the code size considerably without much benefit. Resolves #139.

Language identifiers like LangID::EN_US have a new type, but this should not require source code changes in a code which uses them.

Provide two traits to convert LangID to and from u16 freely. This implicitly provides TryFrom trait which previously was derived using TryFromPrimitive.

There's no special constructor for LangID for values out [... cut]
commit b96ccb91 (2024-02-28): Convert LangID from u16 enum to tuple struct containing u16

Relax lang_id checks provided by enum so that any lang_id could be
used, not only values listed in the enum. These checks increased
the code size considerably without much benefit. Resolves #139.

Language identifiers like LangID::EN_US have a new type, but this
should not require source code changes in a code which uses them.

Provide two traits to convert LangID to and from u16 freely. This
implicitly provides TryFrom trait which previously was derived
using TryFromPrimitive.

There's no special constructor for LangID for values ou [... cut]
closed 2024-02-28 by ryan-summers